chuot/rdio-scanner #314: Talkgroups Dissapear

hoffadg, 2023-01-13:
After importing ~800 radioreference talkgroups and running for a while, sometimes all of the talkgroups for that system will eventually disappear.  I've seen it happen a few times over the last couple of weeks on on both the Windows and Linux version.

chuot, 2023-01-29:
Is there any Rdio Scanner logs when that happens?

It happens once for a user where its database was corrupted. Again logs will help to troubleshoot.

hoffadg, 2023-02-01:
> Is there any Rdio Scanner logs when that happens?
> 
> It happens once for a user where its database was corrupted. Again logs will help to troubleshoot.

I will grab the logs the next time it happens.  It hasn't happened in the last few weeks.

chuot, 2023-03-12:
Closing since no new info on this.

closed 2023-03-12 by chuot