chuot/rdio-scanner #321: Write constraint failed error

cpg178-kcd, 2023-02-03:
Recently setup a 2nd server running Rdio and configured a downstream to send the audio files to the main server, occasionally I am getting a 
controller.ingestcall: systems.write: constraint failed: UNIQUE constraint failed: rdioScannerSystems.id (2067) error.

On this setup I am using the default sql lite DB so my suspicion is that the main RDIO instance is already trying to write something at the same time that the downstream it hits so it errors out, or am I missing something here?

chuot, 2023-02-05:
It tries to create a new system with an ID that already exists. Is auto-populate activated on that system or globally?

cpg178-kcd, 2023-02-06:
Yes, the error seems to have stopped since the system has been up and running for a few days now. i will close this

closed 2023-02-06 by cpg178-kcd