chuot/rdio-scanner #322: Unable to playback audio on mobile app iOS

cpg178-kcd, 2023-02-03:
Not sure if related to my previous issue but after enabling the downstream, when I connect to the server that is ingesting the audio sent by the secondary server it just skips through any audio files that came from the upstream instance, but anything recorded locally to the server plays fine. This happens on the iOS mobile app, iOS safari browser, but the audio plays back fine on a desktop computer via chrome browser so the audio files got ingested fine but refuse to playback. There is no errors in the logs regarding the playback issues either.

chuot, 2023-02-05:
Are you using the audio conversion feature? This makes audio files a bit more compatible with mobile devices. Also, for audio conversion to work, FFMPEG must be installed on the same machine that is running Rdio Scanner.

cpg178-kcd, 2023-02-06:
I do not, I will have to give that a try. It is interesting since files coming from SDRTrunk are also MP3 but they work fine, but if the MP3 comes from proscan it does not play on an apple device. I have ruled the downstream out of it as it behaves the same if I connect directly to the satellite server that is collecting these audio files.

cpg178-kcd, 2023-02-06:
Installing FFMPEG and rebooting RDIO fixed the issue. Closing

closed 2023-02-06 by cpg178-kcd