chuot/rdio-scanner #330: Downstream timing out

cpg178-kcd, 2023-02-27:
Out of the blue I am constantly getting a "connectex: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond."

I have rebooted the internet connection, and servers on both ends. It will successfully pass 1 file to the downstream but then everything else after fails. Nothing has changed on either device.

cpg178-kcd, 2023-02-27:
Did some sniffing with wireshark and there is clear communication between both devices, but what is interesting is every packet appears to be UDP, whereas the error keeps saying "Dial TCP" and rarely "read TCP"

cpg178-kcd, 2023-02-27:
[image]

chuot, 2023-02-27:
The context expiration time is 30 seconds. This could be due to many factors such as a problematic proxy between the two, insufficient systems resources, problem with the network and so on.

More investigation is needed. Look at the system logs for something that could explain this.

cpg178-kcd, 2023-02-27:
So interestingly enough after 4 hours of troubleshooting last night, I gave up for the evening and discovered this morning it is working fine. 

I will look deeper into any network issues, my initial suspicions was the limitations of SQLite trying to ingest multiple files from the very active trunking system along with multiple downstream coming in at the same time, but moving to MariaDB did not make any change.

closed 2023-03-12 by chuot