chuot/rdio-scanner #338: Config does not save and erases TGs at end user

nickd3214, 2023-03-17:
When attempting to add users/access codes, specifically if I'm trying to only allow certain users to access certain systems, the config will not save and in the log I get:
admin.confighandler.put: systems.write: constraint failed: UNIQUE constraint failed: rdioScannerSystems.id (2067)
The only fix is to delete all users and then re add. Anytime I attempt to edit any users access it will not save and erases all channels for that access.

chuot, 2023-03-17:
It looks like you have removed a system, and recreate it right away with the same ID without saving between the two operations. This breaks the constraint on the database table.

This kind of behavior will be fixed with the upcoming v7 where the database structure has been completely revamped.

Until v7 is released, when removing a system, save the config then recreate your system and save again.

nickd3214, 2023-03-17:
Thank you for the quick reply! Looking forward to v7. In meantime, will not altering the actual systems & the IDs likely prevent this issue?

dtnetnt, 2023-03-17:
Hi. I've had the same problem when adding new users. I found that if I added the new user then restarted the rdio-scanner service I didn't get the error that you mentioned. Hope that helps

closed 2023-04-17 by chuot