chuot/rdio-scanner #339: Loss of audio when running Rdio in browser on android

Nirbles, 2023-03-23:
Hello: I have an issue when I run rdio in a browser on my Samsung 23+, specifically the audio stops a short time after I start it and put my phone on my belt (no buttons are being pressed). The program is still running but the audio won't come back without refreshing. I have designated the browsers as never sleep and turned off battery saver settings I can find. Any ideas, or is their a way to run two systems in the app would be appreciated as a workaround. Thank you.

chuot, 2023-03-25:
This may be because you have not enabled audio conversion on the server. Playing certain audio file formats may be problematic on some mobile devices. Audio conversion requires the FFMpeg software to be installed on the same host as the Rdio Scanner server. Make sure it is installed, then check the audio conversion option on the admin dashboard.

Nirbles, 2023-03-25:
Thank you for reply. I checked...currently the setting is "enabled without normalization" should I change it?

chuot, 2023-03-25:
No, you want it enabled. But is FFMpeg installed on the PC and does the PATH environment variable point to it?

If it's not installed, you should get a message saying "ffmpeg is not available, no audio conversion will be performed" when you start the server and when it first tries to convert an audio file.

If you are running Rdio Scanner on a Windows PC, I suggest you read the FAQ section on the PDF document that comes with the release. Unfortunately, installing FFMpeg on Windows isn't as easy as it is on a Linux machine.

Nirbles, 2023-03-25:
I turned it off and on and saw no error message. I can also select with normalization or with loud normalization.

Nirbles, 2023-03-25:
I verified FFMpeg is installed for Audacity. I do not see anything about this in the FAQ on page 19. How do I check the path environment?

wmg819, 2023-03-25:
Start at step 12 here https://www.wikihow.com/Install-FFmpeg-on-Windows this link is from the rdio-scanner pdf (FAQ pg 20) and will show you how to check the environment path.

Nirbles, 2023-03-25:
I followed all the steps and set the environment... when I run command I get:
'ffmpeg' is not recognized as an internal or external command,
operable program or batch file.

wmg819, 2023-03-25:
First is this on a Windows system? If so do you have the ffmpeg folder on your C: drive or is it in another location or folder? When you open a command prompt and type ffmpeg -version does it list it? If your running a Linux Distro the set up would be different.

Nirbles, 2023-03-25:
Yes Windows 11 and yes in C:\ drive
[image]
[image]
[image]
[image]

Please see attached screen shots...thank you.

Nirbles, 2023-03-25:
Looks like it decided to work:
[image]

Nirbles, 2023-03-25:
now I just seen to set to rdio?

wmg819, 2023-03-25:
Now all you should need to do is start rdio-scanner and go into the admin page, click on config then on options and set the conversion option you want. Once done the window cmd that rdio-scanner is running in should show the calls being ingested with the .m4a extension.

Nirbles, 2023-03-25:
Thank you...is there any difference between With normal, without normal and/or with loud?

wmg819, 2023-03-25:
For the 2 systems I monitor enabled without normalization seems to work well but I suggest you try each one to see how it sounds to you on your phone.

Nirbles, 2023-03-25:
Success! (I will try the phone again)
[image]

closed 2023-04-17 by chuot