chuot/rdio-scanner #341: Feature request:  Freeze talkgroup name

Clark985, 2023-04-05:
[low priority as it's just a cosmetic thing]

I'm running a main instance of RDIO-Scanner+SDRTrunk along with several other instances pushing their calls to me with the downstream feature (I'm capturing a good chunk of the statewide system in West Central Minnesota).  

Some of the other instances that I don't have direct control over have their talkgroups named differently in SDRTrunk.  The last system to push a call to a given talkgroup will set the name, causing the names to flip back and forth.

Disabling "auto populate" would probably remedy this globally, but then new talkgroups would be missed.  

Keep up the great work!  Looking forward to v7.

Clark985, 2023-04-05:
Disregard, looks like this was already inquired about in   discussion 279

 https://github.com/chuot/rdio-scanner/discussions/279

tadscottsmith, 2023-04-05:
Curious if you'd be interested in sharing a link to what your capturing? I grew up in the area and would be interested in listening in occasionally!

Clark985, 2023-04-05:
> Curious if you'd be interested in sharing a link to what your capturing? I grew up in the area and would be interested in listening in occasionally!

Can you DM me?  Not seeing an obvious way do to that in here.

chuot, 2023-04-17:
Unfortunately, it's one or the other, auto-populate or not.

closed 2023-04-17 by chuot