chuot/rdio-scanner #342: Please enable javascript error

Bobabadabouskie, 2023-04-13:
I don't usually use the web interface to use rdio but recently I needed to do some admin work. I tried to bring up both the regular page and admin page and the page loads but is blank. When I view page source or inspect the page all I see is "please enable javascript to continue using this application". This is using multiple browsers and devices. 

The android app works fine. The logs don't seem to show anything wrong.

I don't know enough to troubleshoot correctly...aside from making sure that javascript is enabled and stopping and starting rdio again. 

Using the latest version of rdio and debian version 12.

Any help would be greatly appreciated.

Thanks.

chuot, 2023-04-17:
Do you use any sort of a plugin that disable javascipt (like noscript)? Is Javascript enabled for the site browserwise?

Bobabadabouskie, 2023-04-25:
After some troubleshooting and other odd errors I believe my issue was failing ram. Closing this topic. Thank you.

closed 2023-05-26 by chuot