chuot/rdio-scanner #343: Incorrect Hold TG behavior after PC returns from sleep

hoffadg, 2023-04-13:
Running with the latest version on Windows, If I have "Hold TG" enabled when my PC goes to sleep, upon returning if I then disable "Hold TG", all of the TGs remain disabled.  The only way to recover is by manually re-enabling all TGs.  But then you are forced to enable TGs that had been previously disabled individually so it is not an ideal workaround.

chuot, 2023-04-17:
This will be fixed in v7.

tadscottsmith, 2023-07-10:
I occasionally have this issue when leaving "Hold TG" on for too long, even though my PC never sleeps. Is this likely the same issue that you're already addressing?

doftheworld, 2025-05-02:
In the meantime, I had this problem several times and was able to simplify the fix by doing the following in Chrome: 
Go to setting - more tools - developmer tools
Navigate to Storage - Local Storage - the URL for your scanner
Copy the top value on the right hand side - paste that into a notepad file
Then next time it resets so you are only listening to the single talkgroup, go to the same location in developer tools, right  click on that top value on the right side and click "edit value" then cut/paste the data from the notepad file. 

After this close the developer tools, refresh the window with the scanner and it will be like it was before. 

My hope/dream is that a future version will include the ability to save several "Setups" for a customized listening experience that is easily switched between. In the meantime, this is the way to do it.

tadscottsmith, 2025-05-02:
That's what I'm currently doing. I think I gave you that workaround :) It just requires me to remember to resave the config after every time I toggle the value of a TG, so I'd like to get a real fix. 

You can kind of do what you're looking to do by appending an id to the end of the URL. For instance, you can have the following and have different TGs toggled on and off for each one. You'll actually see additional lines in the Chrome local storage for each. There's no easy way to toggle between instances from the GUI, so you have to manually adjust the URL or save them as different bookmarks.

http://localhost:3000/?id=police
http://localhost:3000/?id=fire
http://localhost:3000/?id=ems

https://github.com/chuot/rdio-scanner/blob/master/docs/faq.md

doftheworld, 2025-05-02:
Hahaha and here I thought I was passing along helpful information to someone else!  Thank you for that btw - I've used it countless times. 

Thanks for the idea on how to make instances, that's amazing!