chuot/rdio-scanner #359: Setup Server credentials

Walt1980, 2023-06-16:
HI, I just installed the App on my Android Phone. With the app i can listen the frequecies about near Montreal Canada, but what can I select other countries? I must select the other URL and access code But Where I can find this Url?
thanks for your answer
Walter

ciscoladder, 2023-09-13:
I have the same question. Where do we find SDR servers for our local areas ? I'm in Pennsylvania and would love to have the server url to Chester County and Delaware County.

quester2011-cyber, 2025-09-24:
I have the same question. Where do we find SDR servers for our local areas ?

quester2011-cyber, 2025-09-24:
There is no central list of "rdio scanner" servers because Rdio Scanner is software for creating your own scanner, not a service that provides a list of public servers. You can find online scanner feeds through specific community-run sites like [Alertpage](https://www.alertpage.net/rdio-preview/) and [rdio.alertpagesdr.com](https://rdio.alertpagesdr.com/members/scanners.php), or you can use search terms like "[your city/town name] radio scanner" or "[your city/town name] police radio" to find local feeds.