chuot/rdio-scanner #364: Android app: Error 21 cannot create file when saving call from history

canuckcam, 2023-07-05:
On my Samsung S23 and S21, I get an error of FileSystemException: Cannot create file, path = '/storage/emulated/0/Download' (OS Error: Is a directory, errno = 21) when saving a call from the Search Call section. My source is SDRTrunk 0.6.0 a6 using its built-in API uploader functionality. Server is v6.6.3 on Ubuntu 22.04 LTS.

[image]

Strangely enough I recently noticed the audio files are downloaded as unknown.dat on my Windows 10 laptop (works fine when manually renamed), but the proper filename (.m4a) does show on my Mac!

repairfox, 2023-08-02:
Adding to this, same issue on my Iphone native app. If it makes any difference, I have the paid version. I’m attaching a screenshot. When I access by browser, I can download but have the same .dat extension. However if I change the extension to .mp3 (which is the original format), it works fine. 
[image]

canuckcam, 2023-08-07:
I also have the paid version of the app. Is your source from SDRTrunk a7's Streaming tab? I think it's something to do with how SDRTrunk accesses the API. I might need to go back to running Rdio-Scanner as a client on those machines. My logs show Trunk-Recorder ingests correctly with a filename.

Trunk-Recorder:
08/07 00:30 newcall: system=10 talkgroup=36176 file=36176-1691382647_142290000.0-call_331570.m4a success

SDRTrunk:
08/07 00:30 newcall: system=1 talkgroup=1605 file= success

repairfox, 2023-09-09:
@canuckcam I added ffmpeg today, and now this problem is resolved for me.

canuckcam, 2023-09-09:
Thanks! So I installed ffmpeg, then enabled Options>Audio Conversion: Enabled with normalization. Now I get an m4a.mp4 file. If I disable audio conversion, I see the unknown.dat again.