chuot/rdio-scanner #366: Database error when adding users from command line

kcwebby, 2023-07-15:
Getting a database error when adding users from command line.

from syslog:
```Jul 15 14:30:50 localhost Rdio Scanner[2803]: 2023/07/15 14:30:50 admin.useraddhandler.post: accesses.write: sql: converting argument $7 type: unsupported type []interface {}, a slice of interface```

command run:
```rdio-scanner -cmd user-add +ident username +url http://10.1.100.79  +code "P@ssword" ```

Returned:
```417 Expectation Failed```

kcwebby, 2023-10-12:
This is still occuring on a regular basis;  Eventually, this will have the RDIO instance stop saving configuration changes (users added via command line) to the database, and will require a delete/reinstall of the RDIO scanner with a purge of the database.  This is happening with MySQL being used as the backend.

kcwebby, 2023-10-12:
Suspect it might be due to this being in the config (when exported from console):
``` {
      "_id": null,
      "code": "thisisapasswordtest",
      "expiration": null,
      "ident": "testuser",
      "limit": null,
      "order": null,
      "systems": "*"
    },
    
    ```

chuot, 2023-10-13:
If I recall correctly, _id: null means a new user when using the cmdline tool, but if it came back from the server with a null value, then that's a problem because it shouldn't. Any strange syslogs about this?

kcwebby, 2023-10-13:
We figured this out -- this happens after a duplicate access code is attempted to be inserted by the command line tool.
From that point forward, the rdio-scanner instance has to be rebuilt from scratch, as all users from that point will be _id: null.
and upon restarting, all of those credentials will not be maintained upon a service restart.

``` Oct 12 17:50:57 rdio-instance6 Rdio Scanner [16962]: 2023/10/12 17:50:57 admin.useraddhandler.post: accesses.write: Error 1062: Duplicate entry 'password1234' for key 'rdioScannerAccesses.code'  ```

renfrewcountyscanner, 2023-10-14:
I'm starting to get into the creating IDs via the command line too, just using email addresses as the ID and code.
I'm using 
./rdio-scanner -cmd user-add +ident RCS-SUB_email@domain.com +code email@domain.com +systems 100,200,300 +url http://localhost:3001

And I am getting back
417 Expectation Failed
And the logs show
admin.useraddhandler.post: accesses.write: sql: converting argument $7 type: unsupported type []interface {}, a slice of interface

but it is still creating the logins.  I'm using the default DB that rdio comes with.  v6.6.3

kcwebby, 2023-10-17:
Here's a clean log entry showing what happens for the database error:  Hopefully this makes it clear what is needed?

from my script to add the user:
```
2023-10-17 09:25:01am:7236       Add:username4857:passw0rd@1:/nfs/rdio-exe/rdio-scanner -cmd user-add +ident username4857 +url http://127.0.0.1:3003  +code "passw0rd@1":/nfs/rdio-exe/rdio-scanner -cmd user-add +ident username4857 +url http://127.0.0.1:3003  +code "passw0rd@1":417 Expectation Failed
```

from the RDIO log:
```
10/17
09:25
admin.useraddhandler.post: accesses.write: sql: converting argument $7 type: unsupported type []interface {}, a slice of interface
```

from the exported config after:
```
{7}
_id	:	null
code	:	passw0rd@1
expiration	:	null
ident	:	username4857
limit	:	null
order	:	null
systems	:	*
```

tadscottsmith, 2023-10-17:
I don't know if this is directly related, but one thing worth mentioning, is it seems maybe you're trying to use the `ident` as the unique identifier, when in fact it appears the `code` is the unique identifier in the application. The documentation on the webapp identifies the `ident` as the **Dummy identifier which serves only to identify this access code is for whom.**

For instance, running the following two commands does not actually create two users.

`[radio@rdio-server rdio-scanner-test]$ ./rdio-scanner -cmd user-add +ident username1 +code "password1"`
`User username1 added.`
`[radio@rdio-server rdio-scanner-test]$ ./rdio-scanner -cmd user-add +ident username2 +code "password1"`
`User username2 added.`

What really happens is that the identity tied to existing code of `password1` is updated to be `username2` and `username1` no longer exists. I could see there being issues with a script trying to create a bunch of identities with the same access code.

kcwebby, 2023-10-17:
> I don't know if this is directly related, but one thing worth mentioning, is it seems maybe you're trying to use the `ident` as the unique identifier, when in fact it appears the `code` is the [...]

Yes -- Agreed here;  We are using it because the source of their password has a username, we are just carrying it through to make it easier to find users for troubleshooting/updating passwords, etc.  We have recently run into the duplicate password issue and are making some updates to prevent that.   

In your example, the second command *SHOULD* generate an error for duplicate key.
(maybe not, you are using sqlite, and I am using MySQL)
Rdio Scanner [16962]: 2023/10/12 17:50:57 admin.useraddhandler.post: accesses.write: Error 1062: Duplicate entry 'password1234' for key 'rdioScannerAccesses.code'

tadscottsmith, 2023-10-17:
Any chance that the user before the error always has a `+systems` defined? In my testing it seems like I can create as many users as I want until I add a user with a `+systems` explicitly defined, and then can't create any more.

renfrewcountyscanner, 2023-10-18:
I thought the same,  for me +systems does not work, none of my CLI created users have systems assigned to them.
I did try your idea and without systems, same errors above.  
But it creates users with all systems.

tadscottsmith, 2023-10-18:
What I found is that after you try to create someone with a system once, none of your additional users will work until you rebuild. It's like it puts the JSON into the system field with special characters escaped, and then can't properly read it when it attempts to add another user.