chuot/rdio-scanner #369: Saved configuration changes do not survive a service restart

repairfox, 2023-08-03:
Running on Windows 10 64 Bit, installed as a service. Any changes made in the configuration log this error:
`admin.confighandler.put: systems.write: constraint failed: UNIQUE constraint failed: rdioScannerSystems.id (2067)`

I can export the configuration after every change, then upon a restart I can import the configuration file fine. Things I plan on trying: Uninstalling the service, see if changes save then. Also, making sure the service is running as administrator.

ramarco, 2024-02-18:
I have the same issue and am running on Windows 10 64 Bit, but not installed as a service.
Please let me know if you've found a workaround @repairfox

repairfox, 2024-02-18:
I did. I started a Ubuntu server with linode to run rdio scanner. Since doing that, no trouble with it saving config.

ramarco, 2024-02-18:
Thank you, @repairfox. I'll keep that on the back burner but, for now, I'll see if it gets fixed. I found a post from someone having a similar problem and @chuot said it would be corrected in version 7.