chuot/rdio-scanner #371: 'no system' error on Dirwatch for trunk-recorder

DarthXoc, 2023-08-04:
# Summary
After setting up a Dirwatch on trunk-recorder's capture directory, Rdio Scanner begins displaying errors saying "dirwatch.ingest: no system".

# rdio-scanner config
Version: 6.6.3
## Dirwatch
[image]

## System (Frisco)
[image]

## System (Denton)
[image]

# trunk-recorder
Version: 4.4.0

## Config
Including trunk-recorder config below. System is operational, but I wanted to include it for reference-
```
{
        "ver": 2,
        "sources": [{
                "center": 855337500.0,
                "rate": 10000000,
                "gain": 49.6,
                "ppm": 0.0,
                "digitalRecorders": 12,
                "driver": "osmosdr",
                "device": "airspy=0x0"
        },{
                "center": 771568750.0,
                "rate": 10000000,
                "gain": 49.6,
                "ppm": 0.0,
                "digitalRecorders": 6,
                "driver": "osmosdr",
                "device": "airspy=0x0"
        }],
        "systems": [{
                "control_channels": [858187500,858612500,859037500,859612500],
                "type": "p25",
                "talkgroupsFile": "/home/jason/radio/configs/trunk-recorder/frisco-channel-list.csv",
                "recordUnknown": false,
                "shortName": "Frisco",
                "callLog": true,
                "hideEncrypted": true,
                "hideUnknownTalkgroups": true,
                "modulation": "qpsk",
                "broadcastifyApiKey": "no-key-for-you",
                "broadcastifySystemId": "no-id-for-you"
        },{
                "control_channels": [772606250,773656250,858287500,859287500],
                "type": "p25",
                "talkgroupsFile": "/home/jason/radio/configs/trunk-recorder/denton-channel-list.csv",
                "recordUnknown": false,
                "shortName": "Denton",
                "callLog": true,
                "hideEncrypted": true,
                "hideUnknownTalkgroups": true,
                "modulation": "qpsk",
                "broadcastifyApiKey": "no-key-for-you",
                "broadcastifySystemId": "no-id-for-you"
        }],
        "broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
        "frequencyFormat": "mhz",
        "captureDir": "/home/jason/radio/audio/trunk-recorder",
        "controlWarnRate": "10",
        "callTimeout": "1",
        "audioStreaming": true,
        "plugins": [{
                "name":"simplestream",
                "library":"libsimplestream.so",
                "streams":[{
                        "TGID": 102,
                        "address": "127.0.0.1",
                        "port": 9125,
                        "sendTGID": false,
                        "shortName": "Frisco",
                        "useTCP": true
                },{
                        "TGID": 425,
                        "address": "127.0.0.1",
                        "port": 9125,
                        "sendTGID": false,
                        "shortName": "Frisco",
                        "useTCP": true
                },{
                        "TGID": 533,
                        "address": "127.0.0.1",
                        "port": 9125,
                        "sendTGID": false,
                        "shortName": "Frisco",
                        "useTCP": true
                }]
        }]
}
```

# Example
```
2023/08/04 15:56:02 dirwatch.ingest: no system, /home/jason/radio/audio/trunk-recorder/Frisco/2023/8/4/2-1691182546_854212500-call_24.json
```
(forgive the file structure, that'll be getting cleaned up when I'm done building my next scanner)

jcummingsg2, 2023-08-12:
I just added rdio-scanner along side my existing trunk-recorder server (Ubuntu Server 20.04) and configured rdio-scanner for DirWatch.  Seeing the same behavior described here.

Log file during parsing indicates:
2023/08/12 18:06:06 dirwatch.ingest: no system, /home/n7qob/psern/2023/8/12/2327-1691863556_853525000.1-call_60917.json

Contents of JSON file:
cat /home/n7qob/psern/2023/8/12/2327-1691863556_853525000.1-call_60917.json
{
"freq": 853525000,
"start_time": 1691863557,
"stop_time": 1691863557,
"emergency": 0,
"priority": 4,
"mode": 0,
"duplex": 0,
"encrypted": 0,
"call_length": 0,
"talkgroup": 2327,
"talkgroup_tag": "RetPD Pri",
"talkgroup_description": "Police Primary",
"talkgroup_group_tag": "Law Dispatch",
"talkgroup_group": "Renton",
"audio_type": "digital tdma",
"short_name": "psern",
"freqList": [ {"freq": 853525000, "time": 1691863557, "pos": 0.00, "len": 0.44, "error_count": "0", "spike_count": "0"} ],
"srcList": [ {"src": 12983, "time": 1691863557, "pos": 0.00, "emergency": 0, "signal_system": "", "tag": ""} ]
}

Does it need to parse system from the file name format or from the JSON file.  I don't see system in either.  This is a P25 trunked setup.  I do see a param signal_system under srcList but its null.   

Latest release version of both trunk-recorder (4.6.0) and rdio-scanner (6.6.3)

Neo-CTC, 2024-10-19:
You need to manually edit the database and add a system ID to the directory watch. 

1. Open the `rdio-scanner.db `file with a SQLite editor
2. Open the `rdioScannerDirWatches` table
3. Change the `systemId `column from NULL to any number