chuot/rdio-scanner #375: Can't save changes

Clats97, 2023-08-21:
Hello, I was editing the settings and talkgroups on my browser and I cannot save changes. The save button stays grey no matter what I do. I've tried restarting everything etc and the basic troubleshooting procedures but I cannot save changes anymore. It remains greyed out as if I made no changes. Any ideas?

repairfox, 2023-08-27:
Make sure all required fields in all menus are satisfied. If a required field is empty or has invalid data it won’t save.

aegershman, 2023-11-24:
FYI had the same issue when trying to save settings. "Save" was greyed out. It was due to misconfigured talkgroups. There was a red exclamation mark (!) next to the config items that needed attention.