chuot/rdio-scanner #387: SSL Auto Cert not listening

pknight56, 2023-10-11:
Hello!

I just got a domain for my RDIO setup (running on amazon EC2) and am now ready to deploy the SSL feature to use it.
My DNS records are set properly (with just the A record), and the scanner can be accessed by using "http://myscannerurl.com:3000"
Obviously, I just want to be able to use myscannerurl.com with no ports or http

My startup is this:
`rdio-scanner -listen :3000 -ssl_listen :3001 -ssl_auto_cert myscannerurl.com`

And my port forwarding is:
[image]

But, the scanner is still only accessible using "http://myscannerurl.com:3000"
[image]

Any thoughts as to what I may be missing?
Thank you!!

tadscottsmith, 2023-10-11:
Any reason  you're trying to port forward and not just directly listening on 80/443?

`rdio-scanner -listen :80 -ssl_listen :443 -ssl_auto_cert myscannerurl.com`

pknight56, 2023-10-11:
So I removed the port forwarding and am now listening directly

(In my brain, I forgot that I didnt need port 3000 when I overrode the default settings....)

Now, I can access it via just the URL on my PC, but on the IOS app and Safari, it still wont connect.
In safari it says the connection isnt secure (and leads to the Godaddy default page)
And the app just flat out wotn connect

tadscottsmith, 2023-10-11:
Do you have both 80 and 443 open to the internet in your security group? The certificate challenge requires 80 be open.

pknight56, 2023-10-11:
They are open in the security group and in the Windows Firewall on the instance itself as well:

[image]

doftheworld, 2023-10-11:
Is your router forwarding traffic on the incoming port to the computer hosting the program? My issue when I was setting it up was that I could also access it locally, but not remotely. I was able to fix this by telling the router which local IP address traffic on that port should go to.

pknight56, 2023-10-11:
I fixed most of the errors by migrating the nameserves to Amazon Route 53 from GoDaddy. The www. and root domain both work!

But the issue still arises for the SSL cert. 
When calling the auto cert, I can only encrypt one URL. 
In my case I have the root domain, but when using www.rootdomain its not secure. 
And neither URLs are secure on safari?

tadscottsmith, 2023-10-11:
I don't think it is setup to automatically generate a cert with multiple host names. You could try `rdio-scanner -listen :80 -ssl_listen :443 -ssl_auto_cert myscannerurl.com,www.myscannerurl.com` but I wouldn't bank on it working.