chuot/rdio-scanner #389: Android 14 Issues

Forts117, 2023-10-23:
I recently upgraded my phone to Android 14 and I'm now having issues playing in the background with the phone locked. Sometimes it works but most of the time it ends up just going silent. I unlock the phone and have tons of messages queued that immediately start playing. The app doesn't have any battery or power saving restrictions applied to it, and I didn't have this issue with Android 13. I re-installed (it asked me if the app could run in the background) but still the same issues. Anyone else having trouble?

doftheworld, 2023-10-23:
This same issue is also present with iPhone. If my phone locks the messages will play for a short time, then they will stop playing and queue up. I wish it was possible to make the app "never lock" or whatever the proper term is for it - similar to Netflix and Youtube.

Forts117, 2023-10-23:
Yeah that's exactly what I'm seeing as well. Hopefully can be fixed!

RestNPizza, 2023-12-05:
I just updated to Android 14 today and am experiencing the same issues. Have tried a complete reinstall and making sure battery optimization is off. Unfortunately nothing fixes the issue. This is also a problem on the beta version of the native app. Hoping that @chuot can update the native app soon but I know he's been super busy with things in his personal life and also the next version of rdio-scanner.