chuot/rdio-scanner #393: Hung Sessions

renfrewcountyscanner, 2023-11-13:
I've recently moved to creating personal logins for all my users with a limit of 5 connected devices
People are telling me sometimes it is telling them they have too many sessions but I look at the logs and only see one IP

I think Rdio is not clearing sessions properly also the listener count will slowly increase over days of running and sometimes show me 200+ people logged in, I restart servers and that goes back down to 50-60 which is the expected number.

I think this is one in the same.  It was happening on both a barebones install but also in dockers.

chuot, 2023-11-13:
I do have the same exact problem since before v6.6.3. The cause of this is that the logic is flawed on the server side. That is why I'm currently refactoring the whole thing. I takes time however and I'm fully commited to it. As for the webapp, while refactoring, I'm gonna kick out Angular from the project and will revamp the UI, fully canvas based.

One last thing, the server side refactoring will also be easily cloudable (ie scalable). The next major version of Rdio Scanner will be ready for the big league!

And just to prepare some, the next major release will also ditch the dirwatch feature and will be 100% API REST based. Dirwatch wasn't realiable in some cases and also based on a somewhat unmaintained libary.

The project isn't dead, it's just that life has been a bitch with me lately.

doftheworld, 2023-11-17:
> I do have the same exact problem since before v6.6.3. The cause of this is that the logic is flawed on the server side. That is why I'm currently refactoring the whole thing. I takes time however [...]

Interesting to learn that the dirwatch feature will be going away. I use this in conjuction with SDRTrunk, however SDRTrunk recently added API capability for RDIO Scanner. Now I need to figure out how to make it work. I have to admit, the documentation I have seen so far on the subject is pretty much non-existent. Thank you for your continued work on the project - I know you've been going through a lot. I am quite happy with the current setup (I use it just for myself personally). However, switching to the API method and as you describe, "ready for the big league," I may be able to turn this into a commercial venture (which of course I would compensate you for).

renfrewcountyscanner, 2023-11-17:
> Interesting to learn that the dirwatch feature will be going away. I use this in conjuction with SDRTrunk, however SDRTrunk recently added API capability for RDIO Scanner. Now I need to figure out [...]

If they set it up properly you should be able to just point it to same Rdio API URL that everything else uses.

doftheworld, 2023-11-18:
> If they set it up properly you should be able to just point it to same Rdio API URL that everything else uses.

I don't know what this URL is you speak of? I've tried using the localhost, but it fails to connect.

renfrewcountyscanner, 2023-11-18:
> I don't know what this URL is you speak of? I've tried using the localhost, but it fails to connect.

http://localhost:3000/api/call-upload
You will need to make API keys etc..

doftheworld, 2023-11-18:
> http://localhost:3000/api/call-upload You will need to make API keys etc..

I've tried this and all I get is it flipping between "connected" and "temporary broadcast error".