chuot/rdio-scanner #397: Rdio App Remain Active Upon Screen Lock

bryanwhite56, 2023-11-17:
On my iPhone, it seems that the Rdio audio ceases when the screen goes dark, is this by design?
It's not a selling point and darn frustrating.
Can we emulate ScannerRadio and Broadcastify apps? Their audio continues.
Thanx

renfrewcountyscanner, 2023-11-17:
I have alot of users on both Android and Apple and they do not have issues with this, it is a phone permissions issue.  I had one person in the past few days who had to do a full reinstall on his iphone to resolve this issue.

bobfeser, 2023-11-17:
You will have to look at the Battery usage setting on your phone. There is
a setting that turns apps off when the screen goes black to help save your
battery.

bryanwhite56, 2023-11-20:
I’m pretty familiar with the battery and other permissions.  I don’t see
anything amiss.  Would you detail exactly where to look and what I should
set to? Thank you

bobfeser, 2023-11-20:
I use android so I am not familiar with how to set it on iPhone. It will be something like battery optimization or battery usage restriction.  Try looking on YouTube. I will look in the android and see if I can get the exact setting so you can look for that setting.

bryanwhite56, 2023-12-01:
This isn’t a battery setting issue in iPhones.  Thanks for trying

RestNPizza, 2023-12-05:
This is an issue that I am experiencing as well. Just upgraded to android 14 today and that's when the issue started so on android at least, it's related to that version. Tried a fresh install and making sure battery optimization is off and still experiencing the problem. Looks like there are a couple of issues open related to this so fingers crossed the native app will be updated with a fix soon.

bryanwhite56, 2023-12-06:
Thanx.  I’m iOS so different operating system but same underlying issue. Do
upgrades to this app get pushed out via the App Store or do we need to
manually reinstall the app periodically to get the current version of the
app?
Thanx All

RestNPizza, 2023-12-06:
They should be pushed via app store. That's how it is on android at least. I am guessing it's the same on iOS.

bobfeser, 2024-01-11:
I have now updated my Android phone to the latest and this is now an issue. Sorry to [bryanwhite56](https://github.com/bryanwhite56) as I now see what you are dealing with. This is really annoying!! Hope there is a fix soon.

doftheworld, 2024-01-11:
I am not sure if this is a work-around or not for iPhones...but I found that if I'm using Apple Car Play and leave the app running in the background....then it will continue playing after the phone screen goes dark. If I have the app running and on the screen when the screen goes dark then it stops playing after a couple minutes. No idea why having it in the background lets it keep playing when on Car Play, but it does.

bryanwhite56, 2024-01-13:
thanx. I haven't seen any software updates since I installed it approx Oct 1