chuot/rdio-scanner #398: Rdio audio ceases when app not "on top"

bryanwhite56, 2023-11-17:
On my iPhone, it seems that the Rdio audio ceases when I go to reference another app, e.g. email, text
The app audio should continue even when other apps, including audio apps are in use.
I like to listen to music & when Rdio has something to say, sure, it can do so mixing with with the music.
ScannerRadio does this, not sure about Broadcastify apps? 
I prefer Rdio because of the minimal audio delay, the other apps delay 60-300 seconds.
Thanx

renfrewcountyscanner, 2023-11-17:
See other posting for a response from another user.

bryanwhite56, 2023-12-17:
> Yeah. This looks related to issue #389
> <https://github.com/chuot/rdio-scanner/issues/389> and issue #397

Yes, same as 389.  This issue is 397 so this problem is not the same as
that one, it is that one. Thanx

I've never seen an app store upgrade on this app, hopefully soon