chuot/rdio-scanner #404: Duplicate Detection

titothefixer, 2023-12-28:
Does duplicate detection cover transmissions received from a downstream connection?  I'm receiving transmissions on a statewide p25 system.  There is some overlap between the 2 towers I'm monitoring.  Shouldn't duplicate detection catch these overlapping transmission?  If so, what am I missing?

doftheworld, 2023-12-28:
I catch as many as 4 duplicates on towers I monitor (also on a statewide P25 system) - but never hear them more than once. I would also be interested in how it decides which of the 4 received transmissions to serve up (some of the towers have better quality than others). Wish there was a way to prioritize them.

OhioRSN, 2023-12-28:
Are you using the same computer or different computers?

*Michael Chambers - Ohio Radio Scanner Network*
Founder/Manager/Support
***@***.**** ***@***.***>

titothefixer, 2023-12-29:
I'm running it from 2 separate computers. One is at my house the other is at my friend's house in the next county over. We hear some of the same transmissions they're usually sent back and forth and we both hear them twice.

OhioRSN, 2023-12-29:
You need to make sure the computer clocks stay in sync. I had to overcome
this issue by installing a time sync program

*Michael Chambers - Ohio Radio Scanner Network*
Founder/Manager/Support
***@***.**** ***@***.***>