chuot/rdio-scanner #445: Anyone still working on this project?

seth4380, 2024-10-06:
Is anyone still working on this project? Be nice to be able to have support in latest Linux

chris021, 2024-10-08:
I'm still using it and paying for the iOS app. I hope Chrystian finds the time / enthusiasm for it again. It really is a very solid app. so reliable!

seth4380, 2024-10-08:
I’m sure using it to and paying for the app. I had just noticed that it hasn’t been worked on in a few years. Definitely a solid app and loving it. Sent from my iPhone

Dewey3, 2024-10-09:
I believe Chrystian just revisited it last month, so I'm just optimistically waiting while knowing that he has a life first.

chuot, 2024-10-22:
Yes indeed, I am gradually returning to the project after nearly 2 years of dealing with major personal issues and healing psychological wounds. I am now looking forward with the firm intention of continuing this great adventure that is Rdio Scanner. I am currently refactoring all the code to make it more SOLID as well as adding unit tests. It is a lot of background work, but I assure you that I spend a minimum of 20 hours/week on the project. I will push my new refactored code to the v7-wip branch as soon as it is at least functional. Then, I will tackle the webapp where I want to completely remove Angular to replace it with vanilla ecmascript. And finally, take care of all the backlog that is pending. I am omitting the native applications, which will also have to undergo major transformations. Thank you for following me in this adventure, without you I would not have the motivation to continue!

doftheworld, 2024-10-22:
Thank you for the update Christian! I hope you are doing well. I've been using RDIOScanner for my personal use on both my phone and PC for a couple years now. It has really changed the way I monitor things (especially since I live 4000 miles away from the computer that receives everything). Thank you again for your hard work!

Pcpro745, 2024-10-28:
Good to hear! Cant wait. We use it in Minnesota and push about 100,000 calls a day to it.

OhioRSN, 2024-12-09:
Christian, hope you are doing well, thank you for the update!