chuot/rdio-scanner #464: Feature: Export Talkgroup and Units

s3m1s0n1c, 2025-01-24:
Thanks team for this awesome Project.

Would be nice to have the functionality to export TG and Unit data into CSV
This is useful when you wanna make mass changes like change groups or Tags or TG info

Thanks
Sonic

tadscottsmith, 2025-01-28:
There's an Import/Export option in the Admin panel. 

[image]

s3m1s0n1c, 2025-01-28:
@tadscottsmith ,

That function is for importing/exporting configs not for talkgroups as far as I’m aware :)

Thanks Sonic

tadscottsmith, 2025-01-28:
As far as I can tell, it exports the entire config: access codes, API keys, dirwatch config, downstream config, groups, options, systems, talk groups, units, tags, etc.

DrHolbr, 2025-01-28:
Can confirm that the import/export function handles configuration settings,
systems, and talkgroups. I have used it several times successfully

closed 2026-08-15 by s3m1s0n1c