chuot/rdio-scanner #465: How Do You Find The Owner of an rdio Site

peterkga, 2025-01-30:
I use rdio.georgiascan.com but there is no way to notify the owner of the site that it's down. Honestly I can't even remember how I found the URL. Be nice if there is a site with all of the scanner URL's.

kd4e, 2025-01-30:
Please use broadcastify or openmhz for now - same SEGARRN feed.

We're trying to sort out what broke in the rdio feed.

Sorry for the hassle ...

peterkga, 2025-01-30:
@kd4e So you are aware of the issue with rdio. Is this the proper way to ask about that feed or let someone know it's down?

kd4e, 2025-01-31:
> Message ID: ***@***.***>
>
> I thought the api was stored here:

/home/scanner1/trunk-recorder/plugins/rdioscanner_uploader/rdioscanner_uploader.cc 

... but it appears that this just goes out and reads it.

Where is it actually written, please?

I'm guessing something changed and the api has to be refreshed,

but before I take things dow,n and use the rdio admin interface to

create a new one - I need to know where to put it.

Thanks,

David

TNScanner, 2025-02-16:
@kd4e Hey we run a non official Rdio Scanner discord if you wanna join, for Rdio hosts. https://discord.gg/pebyc3Sj2x

peterkga, 2025-04-14:
@kd43, your site is down again. Getting:

[image]