chuot/rdio-scanner #469: SEARCH CALL:- Large number (>100000) of calls makes search unusable

s3m1s0n1c, 2025-02-25:
Hey Team,

I did look in issues for any issues like this one but didn't find any.. So sorry if this issues is already raised..

So I have over 100000 recorded calls in my RDIO instance.

When you go to use the "Search Call" function and select a certain talk group it basically sits there for 5-10-15 minutes and everything else is in rdio is unresponsive until the select query is done.

The Load on the machine seems to rise a little but no where maxing out the CPUs

I'm guessing its a Sqlite3 issue with trying to do a "select from RadioScannerCalls where talkgroup = x" thru 100000 + records..

Is there a better way to construct the sqlitedb with some kinda index to make it faster?

Ive been involved with another project who uses knex with nodejs for better querys
Could one use GORM or Squirrel for GO to make the query's a bit quicker with larger numbers of calls?

Cheers Sonic

KC3DVR, 2025-04-25:
Ive just started having this issue at just over 1.5 million stored recordings. it appears uploads still works but any playback or live calls stops working.

SimonSheehan, 2025-04-26:
Many of us who have larger / busier systems and a large number of recordings have moved to MariaDB instead of SQLite, and found the performance significantly improves and we don't have the issues we often ran in to on SQLite

AnthonyDeFallo, 2025-04-26:
You have a guide on migrating? About to hit 1,000,000 myself

SimonSheehan, 2025-05-01:
> You have a guide on migrating? About to hit 1,000,000 myself

It's fairly straight forward. Install MariaDB, set up a database and user, give it permissions. Start Rdio with the following parameters in either a config file or on the command line arguments. 

`db_type = mariadb
db_host = ipAddressHere
db_name = rdio
db_user = sqlUsernameHere
db_pass = sqlPasswordHere
`