chuot/rdio-scanner #470: Issues on android application

TNScanner, 2025-02-26:
I recently upgraded to an S25 Ultra, but I've noticed that the Rdio app is practically unusable on the device. I'm not sure if it's due to the app being outdated or a rogue ad causing issues. I can open the app and start listening, but shortly after, it freezes, and the UI becomes completely scrambled. I usually have to force close it, though occasionally, it starts working again on its own.

I've attached a short video demonstrating the issue.

[image]
[image]

https://github.com/user-attachments/assets/803090f9-de48-4a68-804c-2d35f9f76c59

TNScanner, 2025-02-27:
> I'm having the same issue with the S25+. I love this app and I listen everyday. I'm glad it is being updated so I can continue to listen and stream effortlessly.  Is there a way to add more Greenwood, SC channels? I'm looking for anything Lander, Self Hospital and EMS. Thank you for continuing to make this an awesome and informative app.

This GitHub repository is only for the software side of Rdio. The person providing the server URL controls which channels are available for listening, as they are the ones feeding the audio. I recommend checking where you got the URL from to see if there are any contact details for them.

TNScanner, 2025-02-28:
Seems like the update has fixed my issues.

someengineer1, 2025-03-09:
The updated app has actually caused problems for me.  Worked quite well before, but now the first time launching it, while showing the orange Rdio Scanner advertisement, it will not connect.  Have to exit and launch it again.  Similarly when a connection is lost (switching from wifi to cellular etc) it doesn't reconnect automatically like it used to, have to exit and open again.  Often even once it is "connected" playing back historical calls they are all just blank and 1 second long, have to exit and go back in for them to work.

This is on stock android on a Google Pixel.

chuot, 2025-03-09:
An open beta is about to be release on the Google App Store. I encourage you to try it. If you're still experiencing the same issue with the latest beta (1.9.6+149), then uninstall the app completely and reinstall the beta. Then try not entering any server address on the latest welcoming slide by just clicking on connect. This will make you connect to my demo server and it should hopefully work. Then go to the settings screen and change the demo server to the one you're not able to connect and see if it works.

UberPlexCa, 2025-03-09:
My older Android tablet got this update, and now the app crashes.  Was this expected for older hardware?

TNScanner, 2025-03-10:
> My older Android tablet got this update, and now the app crashes. Was this expected for older hardware?

Did you try to clear the app cache, or completely uninstall then reinstall to see if it still crashes?

UberPlexCa, 2025-03-10:
> > My older Android tablet got this update, and now the app crashes. Was this expected for older hardware?
> 
> Did you try to clear the app cache, or completely uninstall then reinstall to see if it still crashes?

Yeah, more than a few times. Whatever this 1.9.6 March 4th 2025 update was, it broke it on my old TAB E 8.0 on Android 7.1.1, which has been running for 3-4 years now without issues.

bburkett52, 2025-03-10:
So far latest beta is working with no issue on IOS 18.3.1

chuot, 2025-03-10:
> Yeah, more than a few times. Whatever this 1.9.6 March 4th 2025 update was, it broke it on my old TAB E 8.0 on Android 7.1.1, which has been running for 3-4 years now without issues.

That is strange, because the minimal Android version on which the latest beta should work is Nougat (sdk 24). I will try to push a new beta build within the next 24 hours that will try to remedy your issue. But for sure the minimal version is sdk 24 to keep up with the libraries I'm using.

chuot, 2025-03-10:
> So far latest beta is working with no issue on IOS 18.3.1

Great news! I'll wait for an extra day or two before releasing hat beta on the main channel as I want more IOS feedback.

bobfeser, 2025-03-10:
I am having the exactly the same issue. I have un-installed and re-installed and no change. Running on S22 ultra

Bob

Bobabadabouskie, 2025-03-13:
> Yeah, more than a few times. Whatever this 1.9.6 March 4th 2025 update was, it broke it on my old TAB E 8.0 on Android 7.1.1, which has been running for 3-4 years now without issues.

I am on version 1.9.6 from March 4 also. I had been having a problem where the app loads and appears to be working but no audio, I would have to force close the app.

Now this morning nothing will work to even connect. When I click on the connect button I get no response, almost like button isn't working/registering a touch.

My device is a Samsung Note 20 Ultra, android version 13. I have cleared storage, cache, un-installed and re-installed. 

Anyone have a link to an older apk?

renfrewcountyscanner, 2025-03-14:
I have users with older versions of Android having the same crashing issue. How can we get an old copy of the apk before these recent updates  @chuot   is that possible?

chuot, 2025-03-14:
The best would be someone be able to grab me a log cat from one those old devices and send it to me in DM. I can only test using virtual android devices, which works ok in my tests.