chuot/rdio-scanner #483: Repeating calls on mobile app

DrHolbr, 2025-03-31:
On the mobile app (beta version, pixel 9 pro) it will often get in an odd state where every call repeats multiple times. It is not 100% of the time it does this but once in this state 100% of calls are repeated. I can induce the behavior on my phone by changing servers. I have been able to confirm this by starting with my server and changing to the default server (or any other server) and the calls are duplicating there as well. Calls duplicate at least twice but sometimes 3 or 4 times. I was able to corroborate with at least one other discord user that has experienced this with android, and have had 2 reports from end users of my server that are on iPhone.

On my phone I have tried clearing cache and storage and re-installing but am still able to reproduce this issue. once the issue has started I am able to clear this issue by force stopping the app, relaunching, and not changing servers.

No similar issues or duplicate calls observed while using the web app.

AnthonyDeFallo, 2025-04-07:
I am getting it too but on the prod app. Do you know how to get the beta so I can test it?

doftheworld, 2025-04-07:
I get this *sometimes* when another sound interrupts the call being played. For instance if navigation gives an instruction then sometimes the call is repeated once, or several times if it is interrupted again. Oddly enough, without touching the phone, it will sometimes lower the volume of the call and play the other sound - but not repeat the call. It will also sometimes pause the call. I can't find any rhyme or reason to which it decides to do.

infusopharm, 2025-04-07:
I think I have been able to replicate this. If you go to settings and then click connect, it starts an additional connection and will duplicate the audio.  Seems like every time you hit connect, it will add another duplicate.

tristatescannersux, 2025-04-07:
Only time I have an issue with repeat calls if the TG is linked into an interoperable TG on the system then it plays the TG then plays the Interop channel TG (repeats but diff TG) is this the issue you are having I wonder ??

AnthonyDeFallo, 2025-04-07:
> I think I have been able to replicate this. If you go to settings and then click connect, it starts an additional connection and will duplicate the audio. Seems like every time you hit connect, it will add another duplicate.

I think you’re right. That aligns with how I’ve seen duplicate calls

TNScanner, 2025-04-07:
> I think I have been able to replicate this. If you go to settings and then click connect, it starts an additional connection and will duplicate the audio. Seems like every time you hit connect, it will add another duplicate.

I can confirm this replicates the issue on the beta app for me.

closed 2026-11-16 by github-actions[bot]