chuot/rdio-scanner #498: Import Talkgroups web server does not import emojis properly.

PaTall, 2025-08-15:
[image]

I use iOS emojis in my Talkgroups file.  This help me identify calls easily when searching in 'Search Call"  and when doing general scanning.

To use emojis I need to manually use the web server to edit and insert the emojis as the IMPORT function does not properly handle them (it substitutes a random character from a strange character set).  

It would be nice to have the V7 Talkgroup Import function properly handle emojis.

Thanks for considering.

jddean123, 2026-10-26:
@chuot Mind if I take a stab at this? I recently had to switch from conventional scanning with my old Uniden to a P25 Phase II trunked system and finally put my SDR to use with SDRTrunk and Rdio Scanner. As a fellow developer, I'd like to start helping out where I can!

I had a look and I think I've found the cause. The talkgroup and unit CSV importers read the file with `readAsBinaryString`, which turns each byte into its own character, so multi-byte UTF-8 characters like emojis get split apart. The fix would be client-only: decode the file as UTF-8 (which also drops the byte-order mark Excel adds to "CSV UTF-8" files), with a fallback to Windows-1252 so existing CSVs with accented characters keep working. No server or format changes.

chuot, 2026-10-26:
Thanks a lot for taking the time to investigate this and for offering to help out. I really appreciate it, especially coming from a fellow developer who's actively using SDRTrunk and Rdio Scanner.

Your analysis makes sense, and it certainly sounds like you've tracked down the root cause. Thanks as well for thinking through a backwards-compatible fix path. That's exactly the kind of contribution that helps move things forward.

That said, would you mind holding off for about two weeks? I'm currently on the final stretch of merging a new private v8 branch that I've been working on for the last couple of months. It's a major release with a lot of changes under the hood, along with a completely new admin dashboard and several other new features.

Since there have been quite a few changes in that area, I'd hate for you to spend time implementing and testing a fix against the current codebase only to have to rework it once v8 lands. I think it'll be much more productive if you wait until the new version is merged, then we can revisit it against the latest code and avoid making you do the same work twice.

Thanks again for digging into this and for offering your help. It's genuinely appreciated, and I'd definitely love to have you take another look once v8 is out.

jddean123, 2026-10-26:
@chuot Wait, you've got a v8 hiding somewhere? I just started reading up on v7! If you'd allow it, I'd love to poke around and help get that to the finish line! I've got a variety of environments to test it in (MacOS M2, Windows 10, Raspberry Pi, lots of Docker experience).

chuot, 2026-10-26:
Thanks for the offer! Any help is definitely welcome.

I'm currently wrapping things up and getting ready to squash merge everything into the main branch. Right now we're talking about roughly 1,200 commits, so there's still a bit of cleanup to do before I can put it out there.

My original plan was to just release binaries, but since you're offering to help test, having additional eyes on it will absolutely help stabilize things before a wider release. As you know, AI is great at writing code... and sometimes just as good at writing bugs, so I'm sure a few have managed to sneak through.

Give me another week or two to get things organized. I also wasn't sure whether I wanted to release it as open source after everything that happened recently, but I've made up my mind and I'm moving forward with it.

Thanks again for reaching out. Your offer to help test across multiple environments is genuinely appreciated and will be very valuable.

chuot, 2026-11-15:
@jddean123 v8.0.0-beta.1 has been pushed, and the release candidate is now available for download in the Releases section. I haven't had time to test it on all architectures yet, and since you kindly offered, your feedback would be greatly appreciated. 😉